tnunnink/L5Sharp #1: No LogixContext class.

J-P-Projects, 2022-04-06:
With regard to the usase/example:

var context = new LogixContext("FileName.L5X");

There are only interface classes in the project.

Is the LogixContext class deliberately missing, or else, can you extend the example?

tnunnink, 2022-04-06:
Sorry the documentation is just outdated already. I have been changing a lot of things over the last few months. I plan to go back through and update it and add more in depth information on the project. I'm trying to get to a stopping point first.

The class implementing the ILogixContext was renamed (for the time being) to **L5XContext** and is the namespace L5Sharp.L5X.

So now this would look like this:

`var context = L5XContext.Load("FileName.L5X");`

-or-

`var context = L5XContext.Parse("L5X String Content");`

Hope this helps.

tnunnink, 2022-04-06:
I'm still in the process of changing things, but if you want to get a better idea of where things are at, or how to use it as of now, check out the unit tests in the project **L5Sharp.Querying.Tests**. It will show a lot of the basic querying you can do.

closed 2023-07-24 by tnunnink