tnunnink/L5Sharp #9: Consumed Tag Attributes

aawilliams85, 2024-01-04:
Hey,

Thanks for providing this great project!!  Even as a complete novice at C# I've been able to test some powerful concepts with it.

One thing I'm still trying to figure out, is there a way to Get/Set the Producer and RemoteTag attributes of a Consumed tag as detailed by 1756-RM014B pages 105 and 110?

tnunnink, 2024-01-04:
Thanks! I'm happy to hear it's working for you.

I had up till now looked over to Produce/Consume tag elements. They weren't a high priority for me, but after looking at them in the documentation and exporting some examples, it seemed pretty straight forward to implement. I went ahead and added them to the latest version [0.18.2](https://www.nuget.org/packages/L5Sharp/0.18.2). Take a look and let me know if you have any issues. I added a couple quick tests but nothing extensive.

I know there are probably several things missing from the library that could still be supported. The goal is to eventually have the entire L5X supported. If you find other missing elements/attributes that you need supported just let me know and I'll take a look.

closed 2024-01-05 by tnunnink

aawilliams85, 2024-01-05:
Thanks, I will give this a shot!