tnunnink/L5Sharp #10: Converting ACD files to L5X?

mutigozel, 2024-02-13:
Can I use this library to convert ACD files to L5X?

tnunnink, 2024-02-13:
Unfortunately, no. This library is just an API wrapper over the L5X schema. It requires that you first have the L5X.

It would be nice if Rockwell provided a way to do that programmatically, but I have not found anything on it.

closed 2024-02-14 by tnunnink

tnunnink, 2025-02-15:
This is now possible with the L5Sharp.Logix extension library. You can add this project from NuGet. It will use the native Logix SDK to open and save the file as an L5X and then load the contents of the file, returning an instance of L5X. This of course requires that you have the SDK installed and licensed locally.

mutigozel, 2025-02-16:
Cool. Can you provide a simple example code?

tnunnink, 2025-02-16:
```csharp
var content = ACD.LoadAsync("\pathtofile\project.ACD");
```