tnunnink/L5Sharp #11: Updating rungs for routines

Aadiv1999, 2024-02-16:
I am trying to modify the rungs in the test routine of my main program. I was able to access the "MainProgram" and the "Test" routine using the Find extension. Is there a way to modify the rungs of a routine without using the Content<Rung> method to access the rungs (I cannot write the modified rungs back to the routine)?

Here is what I have currently implemented:
```
Program main = content.Programs.Find("MainProgram");
Routine inputs = main.Routines.Find("Test");
LogixContainer<Rung> rungs = inputs.Content<Rung>();
```

Please let me know if you need any more information.

tnunnink, 2024-02-16:
Hi,

I'm not sure I understand your question fully, but you can just update the rung logic using the `Text` property of the rungs you have retrieved in your example.

```csharp
Program main = content.Programs.Find("MainProgram");
Routine inputs = main.Routines.Find("Test");
LogixContainer<Rung> rungs = inputs.Content<Rung>();

foreach (var rung in rungs)
{
    rung.Text = "XIC(MyInputTag)OTE(MyOutputTag);";
}
```

This is a bad example but demonstrates how to update the logic of a rung if that is indeed what you are trying to do.

If you want to add, remove, insert or update rungs you can also use the LogixContainer (what `Content<Rung>()` returns) like so.

```csharp
rungs.Add(new Rung("Added Rung"));
rungs.RemoveAt(2);
rungs.Insert(0, new Rung("Inserted Rung"));
rungs.Update(r => r.Comment = "This updates all rung comments");
rungs.Update(r => r.Comment = "This updates rung 1 comment", r => r.Number == 1);
```

Also if all you want to do is update properties on a `Rung` object you can get to the rungs maybe easier than you currently are using the Query method.
```csharp
var rungs = content.Query<Rung>().Where(r => r.Container == "MainProgram" && r.Routine?.Name == "Test");

foreach (var rung in rungs)
{
    rung.Text = "Some Updated Logic";
}
```
This won't let you add/remove/insert/update rungs to the collection since it just returns IEnumerable but you can still modify the rung properties like Text or Comment for the rungs in you MainProgram/Test routine this way.

Hopefully I'm answering your question but let me know if not.

Aadiv1999, 2024-02-16:
Thank you for your quick reply, this was immensely helpful, just like this project. Kudos!

closed 2024-02-16 by Aadiv1999