tnunnink/L5Sharp #12: Creating Safety Programs

AndyJRobo, 2024-03-08:
Hi, 

Is there a way to add the class attribute to a program and/or tag to specify it as a safety program/tag?

E.g.
<Program Use="Target" 
    Name="_00s_InMap_CLP330" 
    TestEdits="false" 
    MainRoutineName="Main" 
    Disabled="false" 
    **_Class="Safety"_**
    UseAsFolder="false"> 
 </Programs>

tnunnink, 2024-03-09:
Hey,

This was missing but I went ahead and added it to [0.19.5](https://www.nuget.org/packages/L5Sharp/0.19.5). 

Tag, Program, AddOnInstruction, and Task, now all have this property. Look for the `Class` property of type `ComponentClass` and you should be able to set it to `Standard` or `Safety` now.

AndyJRobo, 2024-03-09:
Awesome thanks for the quick response and update. 
This is a great project that I think I will find very useful.

tnunnink, 2024-03-09:
Thanks! Glad to hear it. Let me know if anything else is missing.

closed 2024-03-09 by tnunnink