tnunnink/L5Sharp #15: RLL Routine Rung Comment Generation

Ryushi5, 2024-03-22:
I have run into an issue when creating a new RLL Routine and adding a commented Rung.

In L5X output, the Comment element is added after the Text element. When importing into Studio 5000 (v33.11), it gives the following error:

`Element <Comment> is in the wrong order.`

If I manually swap the order in the L5X, everything imports correctly. Is there something I can do to re-order the way the rung tags are generated?

tnunnink, 2024-03-22:
Hey,

No, I think this is bug. Thanks for reporting it. I will look to see if I can get it fixed here shortly.

Ryushi5, 2024-03-22:
Thank you sir!

tnunnink, 2024-03-22:
Okay try upgrading to [0.19.6](https://www.nuget.org/packages/L5Sharp/0.19.6) and see if that fixes it. The Comment element should now always appear first before the Text element, regardless of when it is set. I added some tests to make sure.

Ryushi5, 2024-03-22:
Confirmed: running the same generation with version 0.19.6 has the Comment tag first, and was successfully imported into Studio 5000.

Thank you again!

closed 2024-03-22 by Ryushi5