tnunnink/L5Sharp #16: New FBD Routine Example

Ryushi5, 2024-03-27:
I've been using the library quite a bit recently and really appreciate the work you've put into it.

One thing that I've found difficult to figure out is creating a new FBD routine and adding a sheet to it. I am able to open existing sheets and manipulate them, but I haven't quite figured out how to start one from scratch. Any advice?

tnunnink, 2024-03-27:
Yeah, working with FBD is not easy. Modeling them in a simple intuitive way was hard, and to be honest, I still consider the FBD code somewhat of a work in progress. I want to refine it when I get more time, but if something isn't working correctly, I would not be surprised. I have not done extensive testing on creating FBD and importing them into Logix.

But to answer your question, you _should_ be able to create a new `Sheet` and add `Block` and `Wire` objects to it. Here is a simple example.
```csharp
var sheet = new Sheet { Number = 1 };
var id1 = sheet.AddAt(100, 100, Block.IREF("InputTag"));
var id2 = sheet.AddAt(100, 200, Block.IREF(100));
var id3 = sheet.AddAt(200, 100, Block.ADD("Add_Block"));
var id4 = sheet.AddAt(300, 100, Block.OREF("OutputTag"));
sheet.Add(new Wire { FromID = id1, ToID = id3, ToParam = "SourceA" });
sheet.Add(new Wire { FromID = id2, ToID = id3, ToParam = "SourceB" });
sheet.Add(new Wire { FromID = id3, ToID = id4, FromParam = "Dest" });
```
Things to note here: 
- When you call `Add` or `AddAt` you get back the ID of the block that was added (these have to be unique for the import to work so the code will just find the next available id and assign it to the block you add).
- All FBD Blocks can be created using the `Block` class which has static factory methods for creating all the known FBD elements found in Logix (some may be missing or incorrect they have not all been tested).
- You can use the IDs returned by Add/AddAt to add wires to the diagram which connect blocks/pins. You will have to know the defined pin names to make this work.

Obviously, even a simple example like this gets complicated. But once your sheet is all configured, adding it to a routine should be simple. It is crucial to set the correct `RoutineType`, `SheetSize`, and `SheetOrientation` for FBD diagrams for them to be imported.
```csharp
var routine = new Routine(RoutineType.FBD)
{
    Name = "Test",
    SheetSize = SheetSize.A0,
    SheetOrientation = SheetOrientation.Landscape
};

routine.Content<Sheet>().Add(sheet);
```

Hope this helps.

Ryushi5, 2024-03-27:
Aha, I see what I was doing wrong. I wasn't passing the Routine constructor the "RoutineType.FBD", I was trying to change the Type parameter after creating the Routine. When I did that, I wasn't able to add a sheet because there was no FBDContent.

Things are working as expected now, thanks again for your help!

closed 2024-03-27 by Ryushi5

tnunnink, 2024-03-27:
Thanks for pointing that out. I can probably improve that by making it update the child content element when you set Type after constructing the Routine. I'll add it to my list of things to look at.