tnunnink/L5Sharp #17: Problem Creating Array Tags

Ryushi5, 2024-03-27:
I'm running into an issue with generating an array tag. For this example, I'm making a BOOL[100] tag.

For the Tag creation, I am setting the Value to the following:
`Value = ArrayType.New<BOOL>(100)`

It could be that there is a better way to do this, but it seems to work except for one thing. The resulting L5X tag that's exported shows its DataType as:
`... DataType="BOOL[100]" Radix="Decimal" Dimensions="100">`

This does not import into Studio 5000, but works just fine if the DataType is "BOOL" with the Dimensions set at "100".

tnunnink, 2024-03-27:
Thanks for reporting this. I forget why I added the index to the data type name for array types. I think I wanted a way to distinguish non-array types from array types, but it has been a while, so I'm not sure. I need to go back and review this and some other things and do some more testing when I find time.

In the interim I have fixed the issue and published a new version [0.19.7](https://www.nuget.org/packages/L5Sharp/0.19.7) which should write the correct name to the DataType attribute for array tags.

Ryushi5, 2024-03-27:
Thanks again for your help!

Sorry to be blowing you up, but I finally have some time to work on a project I've been looking forward to. Having your library has made it so much easier.

tnunnink, 2024-03-27:
No problem! Glad someone is finding this library useful.

closed 2024-03-28 by tnunnink