tnunnink/L5Sharp #19: Working with SafetyInfo

Ryushi5, 2024-04-17:
I am trying to get a SafetyInfo element added to an L5X, and then add a list of SafetyTagMaps. I think I am able to get the SafetyInfo created with an xElement representing the tag maps, but I'm having difficulty figuring out how to incorporate that SafetyInfo into the L5X.

What is the correct way to incorporate the SafetyInfo object?

tnunnink, 2024-04-17:
SafetyInfo is a property of the Controller component. You can access the Controller from the L5X and then get/set the SafetyInfo property.

```csharp
var content = L5X.Load("MyFile.L5X");
            
var info = content.Controller.SafetyInfo;

content.Controller.SafetyInfo = new SafetyInfo
{
    SafetyLocked = true,
    SafetySignature = "..."
};

content.Save("UpdatedFile.L5X");
```

I have not really tested things like `SafetyInfo`, `Security`, or `RedundancyInfo `for the Controller component so if you have issues let me know.

Ryushi5, 2024-04-17:
Ah, ok I see how that works now. It looks like I'll most likely have to build out the structure as an XElement first and then add it to the controller to get the tag maps.

As an example:
`<SafetyInfo SafetyLocked="false" SignatureRunModeProtect="false" ConfigureSafetyIOAlways="false" SafetyLevel="SIL2/PLd">
	<SafetyTagMap> AST_ESTOP=AST_Safety</SafetyTagMap>
</SafetyInfo>`

tnunnink, 2024-04-17:
Yeah, looks like SafetyInfo is just missing the ability to add tags to the SafetyTagMap element, is that right?

If that is the issue, then I will get that added here at some point, so you don't have to edit the XElement manually.

Ryushi5, 2024-04-17:
Yeah, I think that's the only functionality I can see that would really be useful.

That would be awesome, once again I really appreciate your help!

tnunnink, 2024-04-18:
@Ryushi5 Version 2.1.0 adds `SafetyTagMap` as a property to `SafetyInfo`. It implements `IList<TagName>` which gives it the standard list functionality for tag names.

I added a few tests but nothing substantial. If you have issues let me know.

Ryushi5, 2024-04-23:
Thank you for the updates! I had to rework my code a little to work with 2.0+

It looks like the SafetyInfo tag is generated near the end of the L5X, and when importing, Studio 5000 shows this error:

`Error: Line 16878: Element <SafetyInfo> is in the wrong order.
	RSLogix5000Content/Controller
Complete - 1 error(s), 293 warning(s)`

If I move the SafetyInfo tag to be right after the Controller tag in the L5X: it imports properly.

tnunnink, 2024-04-23:
Thanks for reporting. I noticed something similar with the AOI component.

I'm currently adding some code to the base `LogixElement` to allow derived components to specify their element order and whenever a `Set` method is called, the element will ensure the proper order as required by Studio and defined in the derived class. This will give us a uniform way to handle the ordering of elements and help with preventing these errors.

I'll reply when this update is released.

tnunnink, 2024-04-23:
I have released version 2.2.0. This should take care of element ordering issues. Let me know if this resolves your issues with SafetyInfo.