tnunnink/L5Sharp #20: Issue with DateTime Format in AOIs

jojoguy10, 2024-04-18:
Hi,

When I try to write an AOI with L5Sharp then import it, I get an error in Studio stating:
```
Error: Line 1250: Failed to set the 'CreatedDate' property (The parameter is incorrect.).
	RSLogix5000Content/Controller/AddOnInstructionDefinitions/AddOnInstructionDefinition[@Name="AOI_OB16_Mapping"]

``` 

The raw L5X file shows the following:
```typescript
<AddOnInstructionDefinition Name="AOI_OB16_Mapping" Revision="1.1" ExecutePreScan="false" ExecutePostScan="false" ExecuteEnableInFalse="false" CreatedDate="Thu Apr 18 11:12:14 2024" CreatedBy="ProgramGenerator" EditedDate="Thu Apr 18 11:12:14 2024" EditedBy="ProgramGenerator" IsEncrypted="false" SoftwareRevision="33.0">
``` 

When I look at an L5X file that was generated with Studio, itself, the AOI header is: (notice the datetime format)
```typescript
<AddOnInstructionDefinition Name="AOI_OB16_Mapping" Revision="1.1" ExecutePrescan="false" ExecutePostscan="false" ExecuteEnableInFalse="false" CreatedDate="2019-09-09T18:20:54.492Z" CreatedBy="joe.a.kelly" EditedDate="2024-03-01T18:33:47.092Z" EditedBy="joe.a.kelly" SoftwareRevision="v33.00">
``` 
Looking at the AddOnInstruction class, it looks like it's setting the DateTime string using the SetDateTime method where the format is hard-coded:
```c#
protected void SetDateTime(DateTime? value, string? format = null, [CallerMemberName] string? name = null)
{
    if (string.IsNullOrEmpty(name))
    {
        throw new ArgumentException("Name can not be null or empty", "name");
    }

    if (!value.HasValue)
    {
        Element.Attribute(name)?.Remove();
        return;
    }

    if (format == null)
    {
        format = "ddd MMM d HH:mm:ss yyyy";
    }

    string value2 = value.Value.ToString(format);
    Element.SetAttributeValue(name, value2);
}
``` 

When I correct the DateTime format, the error goes away.  Maybe this is a change from different versions of studio?  I'm using v33.

tnunnink, 2024-04-18:
Hey,

Thanks for reporting this. I think I just overlooked the format for the AOI component. Hadn't spent a ton of time testing it to be honest. 

The `SetDate` method was setup to take different formats from derived classes, and if not provided would default to that format found on the root content element. This is just a matter of passing in the correct format for those properties from the `AddOnInstruction` component. I got this fixed and will push a new version here shortly.

tnunnink, 2024-04-18:
I fixed the date time format for `AddOnInstruction` with version 2.1.0. I tested imported a super basic AOI without any parameters, tags, or routines and it imported without error.

closed 2024-04-18 by tnunnink