tnunnink/L5Sharp #21: AOI Rung Numbers

jojoguy10, 2024-04-18:
I tried importing an AOI (generated by L5Sharp), and Studio is complaining that the Number attribute in the Rung element can't be zero.  I looked at the L5X and, sure enough, the Number attribute doesn't get populated.  This is different behavior than a normal Routine in a Program.  Studio doesn't complain about those not having Rung Numbers, but for some reason, when it comes to AOIs, it cares.

tnunnink, 2024-04-18:
That is interesting. I read that it didn't care about Rung numbers for programs and routines. Didn't know it cared for AOIs.

Have you tried setting the Number property for the rungs you add to the AOI? You should be able to get/set that value as needed.

```csharp
 var rung = new Rung { Number = 1 };
```

Let me know if that doesn't work or if I'm not understanding the issue.

jojoguy10, 2024-04-18:
So, I did some more digging. I just found that when you EXPORT an AOI, it shows the rungs as starting at 0, but when I tried to import, it gave an error.  I tried adding the Number parameter (like your comment), but it still didn't work.  I found that I had to put the Number before the Type attributes.
```typescript
<Rung Number="3" Type="N">
    <Text><![CDATA[XIC(Output_Data.0)OTE(Device.Pt00.Data);]]></Text>
</Rung>
``` 

I ended up just writing my own XElement, but it's very odd how it is different for AOIs vs normal Routines.  In normal Routines, I don't have to put the number in the Rung element, like you mentioned.

tnunnink, 2024-04-18:
Got it. Well, I suppose we could add an initialization for the Number property on the Rung objects so that it will always add that attribute before the Type property. Maybe that would help prevent the error? I don't think it would hurt to always have it there by default.

jojoguy10, 2024-04-18:
I think so.  I just don't get why when I try to import the AOI, it complains about the first rung being Number = 0, but when I export it, the first rung is Number = 0.  Maybe it's just a bug in v33?

tnunnink, 2024-04-18:
That is weird. I just tested exporting and re-importing a test AOI on my sample project which is v32 and that worked fine. I also confirmed that the first Rung number is 0. I'll have to see if we have a v33 installation somewhere I can test on.

tnunnink, 2024-04-18:
I updated the `Number` initialization on `Rung` as mentioned here with version 2.1.0. Gonna leave this issue open until I can test v33 import with rung's have number = 0. I'll get back to you on that.

tnunnink, 2024-04-19:
I noticed I forgot to include the Number initialization for the Rung constructor taking the text argument. I got that fixed and it will be included in the next version. I assume that is what you were using and why you didn't see the Number attribute populated on the XML.

As for this original issue, I tested v33 and v32 exporting an AOI, deleting all the Number attributes from the Rung elements, and reimporting. Studio didn't have any issue with this, and it assigned the rung numbers in the order in which they appear in the document, same as does for programs. 

Here is what the Routines part of my import looked like for reference.

```xml
<Routines>
<Routine Name="Logic" Type="RLL">
<RLLContent>
<Rung Type="N">
<Text>
<![CDATA[XIC(Test)OTL(Latch);]]>
</Text>
</Rung>
<Rung Type="N">
<Text>
<![CDATA[XIC(Test)OTL(Latch);]]>
</Text>
</Rung>
<Rung Type="N">
<Text>
<![CDATA[XIC(Test)OTL(Latch);]]>
</Text>
</Rung>
<Rung Type="N">
<Text>
<![CDATA[XIC(Test)OTL(Latch);]]>
</Text>
</Rung>
</RLLContent>
</Routine>
</Routines>
```

I'm going to close this issue. If you think there is something else missing from this library that would be causing import errors, feel free to submit another issue.

closed 2024-04-19 by tnunnink