tnunnink/L5Sharp #22: Bit Descriptions

jojoguy10, 2024-04-19:
I have a tag that's an INT and I wanted to know if there was already an implementation where I can add descriptions to the individual bits (in the `Comment` element)
```typescript
<Tag Name="DO_03" TagType="Base" DataType="INT" Radix="Decimal" Constant="false" ExternalAccess="Read/Write">
<Comments>
    <Comment Operand=".0"><![CDATA[SAFETY RESET PB ENABLE]]></Comment>
    <Comment Operand=".1"><![CDATA[FRONT GATE LATCH ENABLE]]></Comment>
    <Comment Operand=".2"><![CDATA[LEFT GATE LATCH ENABLE]]></Comment>
    <Comment Operand=".3"><![CDATA[RIGHT GATE LATCH ENABLE]]></Comment>
    <Comment Operand=".4"><![CDATA[OPERATOR WORKLIGHT ENABLE]]></Comment>
    <Comment Operand=".5"><![CDATA[SPARE]]></Comment>
    <Comment Operand=".6"><![CDATA[BOWL FEEDER 1 ENABLE(SMALL RIVETS)]]></Comment>
    <Comment Operand=".7"><![CDATA[BOWL FEEDER 2 ENABLE(LARGE RIVETS)]]></Comment>
    <Comment Operand=".8"><![CDATA[LIGHT CURTAIN ZONE INDICATOR LIGHT GREEN]]></Comment>
    <Comment Operand=".9"><![CDATA[LIGHT CURTAIN ZONE INDICATOR LIGHT RED]]></Comment>
    <Comment Operand=".10"><![CDATA[SPARE]]></Comment>
    <Comment Operand=".11"><![CDATA[SPARE]]></Comment>
    <Comment Operand=".12"><![CDATA[SPARE]]></Comment>
    <Comment Operand=".13"><![CDATA[SPARE]]></Comment>
    <Comment Operand=".14"><![CDATA[SPARE]]></Comment>
    <Comment Operand=".15"><![CDATA[SPARE]]></Comment>
</Comments>
<Data Format="L5K"><![CDATA[0]]></Data>
  <Data Format="Decorated">
    <DataValue DataType="INT" Radix="Decimal" Value="0"/>
  </Data>
</Tag>
```

tnunnink, 2024-04-19:
Currently no way to do that, but I'm sure something can be added. I'll look into it when I get time.

tnunnink, 2024-04-23:
With version 2.2.0 you can access the `Comments` collection for any `Tag`. With this you can add/remove/update `Comment` objects as needed, including bit members. 

```csharp
var tag = new Tag("Test", 100);

tag.Comments = new LogixContainer<Comment>
{
    new(".1", "This is the comment"),
    new(".2", "this is another comment")
};

tag.Comments.Add(new Comment(".3", "This is an added comment"));
```

Also, setting `Description` will still update the corresponding comment for nested tag member objects, as it did before. We just don't have access to bit members as nested tag members.

closed 2024-04-23 by tnunnink