tnunnink/L5Sharp #26: Parsing to a DataType

jojoguy10, 2024-05-03:
Related to my other feat7ure request, is there a function available to parse a string of a `DataType` to a `DataType` instance?

Something like this, but then output to the actual `DataType` ( or rather, `LogixType`) to use in making `Tags`?
```c#
private object GetMemberDataType(Parameter parameter)
{
    object output;
    switch(parameter.DataType)
    {
        case "BOOL":
            output = new BOOL();
            break;
        case "DINT":
            output = new DINT();
            break;
        case "INT":
            output = new INT();
            break;
        case "LINT":
            output = new LINT();
            break;
        case "LREAL":
            output = new LREAL();
            break;
        case "REAL":
            output = new REAL();
            break;
        case "SINT":
            output = new SINT();
            break;
        case "UDINT":
            output = new UDINT();
            break;
        case "UINT":
            output = new UINT();
            break;
        case "ULINT":
            output = new ULINT();
            break;
        case "USINT":
            output = new USINT();
            break;
        case "MESSAGE":
            output = new MESSAGE();
            break;
        case "STRING":
            output = new STRING();
            break;
        case "TIMER":
            output = new TIMER();
            break;
        default:
            output = new BOOL();
            break;
    }

    return output;
}
```

jojoguy10, 2024-05-03:
Or maybe, and this could also be related to #24, can I instantiate a new `Member` with a string as the `LogixData` type?  Granted, that would be require the program to make sure the other properties are configured properly, possibly (unless Logix will find defaults for them when uploaded).

```c#
public Member(string name, string dataType)
    : base(CreateMember(name, data))
{ }
``` 

Seems like that kind of breaks a few things with how you're making the `XElement` :/ .

tnunnink, 2024-05-04:
I agree and I had been thinking about how to do this. I have added a new method in version [2.3.0](https://www.nuget.org/packages/L5Sharp/2.3.0) to create LogixData instance by just providing a name.

```csharp

//Creates a concrete DINT instance.
var atomic = LogixData.Create("DINT");

//Creates a concrete TIMER instance.
var predefined = LogixData.Create("TIMER");

//Created a ComplexData instance with name "MyType"
var data = LogixData.Create("MyType");
```

If the data type name is not known then it resorts to the normal ComplexData instance.

I was going to add a new constructor to Member but it caused too many test errors, since StringData is implicitly convertible to a string type in .NET. But you can just call this method like so.

```csharp
var member = new Member("MemberName", LogixData.Create("TIMER"));
```

This let me add new constructors in a few other places, like ArrrayData and Tag.

```csharp

//Creates 100 DINT objects from the provided name.
var array = new ArrayData<LogixData>("DINT", new Dimensions(100));

//Creates the TIMER instance from the name.
var tag = new Tag("MyTag", "TIMER");
```

closed 2024-05-04 by tnunnink