tnunnink/L5Sharp #28: Adding rung to routine with no rungs

AdamVanSickleAAS, 2024-05-09:
I'm trying to retrieve the collection of Rungs for a Routine that currently has no Rungs defined (it works as I expect for a routine with rungs).  My L5X content was created by importing an L5X document exported from Studio 5000.

When I call routine.Content<Rung>() on a routine with no rungs, it throws the following exception:

`System.InvalidOperationException: 'The required attribute or child element 'RLLContent' does not exist for Routine.'`

Stepping through the code and inspecting the offending routine shows that its Type is set to RLL just prior to making the call that causes the exception.  If I explicitly set the Type to RLL, then the call works as I expect (which I expect is a bug). See the following:

```
public static LogixContainer<Rung> Rungs(this Routine routine) 
{
    LogixContainer<Rung> rungs = null;
    try
    {
        var type = routine.Type;          // This is already set to "RLL"
        rungs = routine.Content<Rung>();  // This throws an exception
    }
    catch (InvalidOperationException) 
    {
        routine.Type = RoutineType.RLL;   // explicitly re-set the type to "RLL"
        rungs = routine.Content<Rung>();  // this works as I'd expect.
        rungs.Add(new Rung("Test Rung")); // as does this
    }
    return rungs;
}
```
I'm guessing explicitly setting the type to RLL causes the underlying RLLContent element to be created for the Routine?
The above seems to be a work around, but is there a preferred way to initialize a collection of Rungs for a Routine?

tnunnink, 2024-05-09:
Thanks for reporting this. Didn't know the Routine content element was not there by default for empty routines. I went ahead and added a fix that ensures an empty content element is created when accessing it via `Content<TCode>()` so that you won't get an exception, and so that we can add Rungs to it as expected. See version [2.3.2](https://www.nuget.org/packages/L5Sharp/2.3.2) for the fix.

AdamVanSickleAAS, 2024-05-09:
Cheers!

closed 2024-05-09 by AdamVanSickleAAS