tnunnink/L5Sharp #29: Feature Request: AOI Rung Instruction

jojoguy10, 2024-05-20:
I was thinking it might not be that difficult to generate the rung instruction neutral text for an AOI.  I was playing around with it, and it seems like it follows the same pattern:
`<AOI Name>(<AOI Tag>, <InOut Tags>, <Input Tags>, <Output Tags>);`

Maybe it's that easy?

tnunnink, 2024-05-20:
Yeah, not too bad. I think only Parameters with `Required` checked are those that are included in the instruction signature. I can add a couple methods to generate a `Instruction` or `NeutralText` object, but all it would do is set the instruction key. You'd have to provide the remaining arguments as either references to tags or immediate atomic values. We can't get that info from the AOI itself.

For example: 
```csharp
var aoi = new AddOnInstruction("TestAOI");
var text = aoi.ToText("TestAoiTag", "Param1", "Param2", 123, 0);
text.Should().Be("TestAOI(TestAoiTag,Param1,Param2,123,0)");
```

All this would do is call `New()` on the `Instruction` class internally and provide it the name of the AOI.
```csharp
var instruction = Instruction.New("TestAOI", "TestAoiTag", "Param1", "Param2", 123, 0);
instruction.Text.Should().Be("TestAOI(TestAoiTag,Param1,Param2,123,0)");
```

Let me know if that is what you are looking for or not.

tnunnink, 2024-06-03:
[3.0.0](https://www.nuget.org/packages/L5Sharp/3.0.0) adds these methods in case you are interested.

closed 2024-06-03 by tnunnink