tnunnink/L5Sharp #33: [Bug] Missing support to DT/LDT Atomic type

frolra, 2024-09-24:
When I try to import my L5X file, the DataType read process throws me this exception:

_`The name 'DT' does not represent a known L5Sharp.Core.AtomicData type.`_
   _at L5Sharp.Core.AtomicData.Parse(String name, String value)
   at L5Sharp.Core.LogixSerializer.DeserializeAtomic(XElement element)
   at L5Sharp.Core.LogixSerializer.DeserializeData(XElement element)
   at L5Sharp.Core.LogixSerializer.Deserialize(XElement element)
   at L5Sharp.Core.LogixSerializer.Deserialize[TElement](XElement element)
   at L5Sharp.Core.LogixElement.GetData()
   at L5Sharp.Core.Member.GetData()
   at L5Sharp.Core.Member.get_Value()
   at L5Sharp.Core.Tag.get_Value()
   at L5Sharp.Core.Tag.get_DataType()_

I Checked at source code and missing this type definition. From Logix DT is Date/Time format and basically is a Unsigned integer 64 bits 
[image]

tnunnink, 2024-09-24:
Thanks. I'll look into adding support for it.

frolra, 2024-09-24:
I started to create a DT class from ULINT if can help you :)
`using System;

namespace L5Sharp.Core;

/// <summary>
/// Represents a <b>DT</b> Logix atomic data type, or a type analogous to a <see cref="DateTime"/>.
/// </summary>
public sealed class DT : AtomicData, IComparable, IConvertible, ILogixParsable<DT>
{
    /// <summary>
    /// The underlying primitive value which is set upon construction and not changed.
    /// </summary>
    private readonly DateTime _value;

    /// <summary>
    /// Creates a new default <see cref="DT"/> type.
    /// </summary>
    public DT()
    {
    }

    /// <summary>
    /// Creates a new <see cref="DT"/> with the provided value.
    /// </summary>
    /// <param name="value">The value to initialize the type with.</param>
    public DT(DateTime value)
    {
        _value = value;
    }

    /// <summary>
    /// Creates a new <see cref="DT"/> value with the provided radix format.
    /// </summary>
    /// <param name="radix">The <see cref="Core.Radix"/> number format of the value.</param>
    public DT(Radix radix) : base(radix)
    {
    }

    /// <summary>
    /// Creates a new <see cref="DT"/> with the provided value.
    /// </summary>
    /// <param name="value">The value to initialize the type with.</param>
    /// <param name="radix">The optional radix format of the value.</param>
    public DT(DateTime value, Radix radix) : base(radix)
    {
        _value = value;
    }
    
    /// <inheritdoc />
    public override string Name => nameof(DT);

    /// <inheritdoc />
    public int CompareTo(object? obj)
    {
        return obj switch
        {
            null => 1,
            DT typed => _value.CompareTo(typed._value),
            AtomicData atomic => _value.CompareTo((DateTime)Convert.ChangeType(atomic, typeof(DateTime))),
            ValueType value => _value.CompareTo((DateTime)Convert.ChangeType(value, typeof(DateTime))),
            _ => throw new ArgumentException($"Cannot compare logix type {obj.GetType().Name} with {GetType().Name}.")
        };
    }

    /// <inheritdoc />
    public override bool Equals(object? obj)
    {
        return obj switch
        {
[... 179 lines cut ...]
            return new SINT((sbyte)_value.Ticks);
        if (conversionType == typeof(INT))
            return new INT((short)_value.Ticks);
        if (conversionType == typeof(DINT))
            return new DINT((int)_value.Ticks);
        if (conversionType == typeof(LINT))
            return new LINT(_value.Ticks);
        if (conversionType == typeof(REAL))
            return new REAL(_value.Ticks);
        if (conversionType == typeof(LREAL))
            return new LREAL(_value.Ticks);
        if (conversionType == typeof(USINT))
            return new USINT((byte)_value.Ticks);
        if (conversionType == typeof(UINT))
            return new UINT((ushort)_value.Ticks);
        if (conversionType == typeof(UDINT))
            return new UDINT((uint)_value.Ticks);
        if (conversionType == typeof(ULINT))
            return new ULINT((ulong)_value.Ticks);
        
        throw new InvalidCastException($"Cannot convert from {GetType().Name} to {conversionType.Name}.");
    }

    #endregion
}`

tnunnink, 2024-09-24:
Do you want to submit a PR?

frolra, 2024-09-24:
Yep, i will review more deeply the code and i submit the PR 👍🏼