tnunnink/L5Sharp #44: [Bug] Module.Create(string, string, Address?) not implementing address

jojoguy10, 2024-11-19:
I tried using the `Module.Create(string name, string catalogNumber, [Address? address = null])` method, but it looks like the address parameter is not implemented in the constructor:
```
public static Module Create(string name, string catalogNumber, Address? address = null)
{
    CatalogEntry catalogEntry = new ModuleCatalog().Lookup(catalogNumber);
    return new Module
    {
        Name = name,
        CatalogNumber = catalogEntry.CatalogNumber,
        Revision = catalogEntry.Revisions.Max(),
        Vendor = catalogEntry.Vendor,
        ProductType = catalogEntry.ProductType,
        ProductCode = catalogEntry.ProductCode,
        Ports = new LogixContainer<Port>(catalogEntry.Ports.Select((PortInfo p) => new Port
        {
            Id = p.Number,
            Type = p.Type,
            Address = ((p.Type == "Ethernet") ? Address.IP() : Address.Slot(0)),
            Upstream = !p.DownstreamOnly
        }).ToList()),
        Description = catalogEntry.Description
    };
}
```

jojoguy10, 2024-11-19:
Along with that (and I can put in a separate issue if you want), is there a reason the `Module.IP` property is read-only?  It would be really useful to be able to set the IP address of modules.

tnunnink, 2024-11-20:
Good find. I must have just missed this when I created that method. I'll get it implemented for the next update.

IP was sort of a helper property to get the value configured on the Ethernet Port. But I agree that it would be valuable to set in the same way. I can also see about adding that in.

tnunnink, 2024-11-28:
Okay this was fixed for 4.7.2 but have not had the chance to fully test it. If it works for you let me know and I will close this issue.

closed 2025-02-15 by tnunnink