tnunnink/L5Sharp #45: [Enhancement] Safety-Related Functionality

jojoguy10, 2024-11-20:
I really like this library and it's been super helpful, but the thing I keep finding myself struggling with is dealing with safety devices.

Generally, Logix will auto-fill a lot of fields, but there are some fields that don't auto-fill and Logix throws an error when they're not filled in.  Specifically, I'm looking at the `SafetyScript` element of the `Module` element.  I've done some testing, and found that if I have a drive that is a CIP safety drive on a safety controller, it needs to see the `SafetyScript` element or it won't work.

It's located in `Module>Communications` and looks like this:
`<SafetyScript Size="59" />`

I haven't figured out what the Size property means, but at the least, having that as a property of the `Communications` class would be great!

jojoguy10, 2024-11-20:
I don't know what I was doing wrong, but it does look like it's adding the SafetyScript after all.  I wonder if I wasn't enabling the safety properly.  Sorry about that!

jojoguy10, 2024-11-20:
OK, I'm still finding ways I can confuse myself, haha!  I guess I was editing the XML in some deep code and that is what was adding the SafetyScript.  I think this issue is still open, but there are work-arounds that aren't too bad.

tnunnink, 2024-11-20:
Thanks for sharing. To be honest I have not worked with safety devices, and it definitely hasn't been a focus with this project, but we can for sure look at adding it.

It would be helpful if you could share an example L5X export for a safety module so I can examine it and use it to run tests against. 

From the documentation I don't see anything referring to a SafetyScript element, but I take that with a grain of salt because there is a lot in their documentation that is outdated.

[image]

jojoguy10, 2024-11-20:
I also couldn't find anything about SafetyScript in their documentation.  Here is an example L5X file that contains a safety PLC with a bunch of safety-enabled servo drives (module names ending in "DRIVE").
[GeneratedProject.zip](https://github.com/user-attachments/files/17835839/GeneratedProject.zip)

closed 2025-05-04 by tnunnink