tnunnink/L5Sharp #47: Cannot access Tasks or Tags from L5Sharp.Core.Controller (and can't find LogixContent class in VS Object browser)

YouCanCallMeGeorge, 2024-12-10:
This is an excellent library.

I've started by writing a simple program that opens an existing L5X file and creates a TreeView showing the object tree: Controller, Tasks, Programs, Routines.

I list the tags at the Controller level, and for each Program.

---->The primary problem is that the L5Sharp.Core.Controller class does not include a Tasks member, nor does it include a Tags member.
The workaround is to get to Tasks and Tags via L5Sharp.Core.Controller.L5X, as follows:
   L5Sharp.Core.Controller oController;
   foreach (L5Sharp.Core.Task oTask in oController.L5X.Tasks)
   foreach (L5Sharp.Core.Tag oTag in oController.L5X.Tags)
But these lines result in two compiler warnings: Dereference of a possibly null reference.

In the documentation I see references to the LogixContent class, and its Tags member for controller tags. But I cannot find the LogixContent class in the C# object browser. (Presumably, the LogixContent class also contains a Tasks element.)

---->Looking at the object browser, it seems natural to add Tags and Tasks to L5Sharp.Core.Controller.
But of course I would be happy if I could find the LogixContent class and retrieve Tags and Tasks there.

YouCanCallMeGeorge, 2024-12-10:
The (now) obvious solution is to pass the L5X pointer in to the Task and Tag listing routines, rather than the L5X.Controller pointer. This works, but is conceptually awkward. Controller tags and tasks should (conceptually) be accessible from the Controller object.

tnunnink, 2024-12-10:
The documentation is outdated. I haven't had time to update it. LogixContent is now [L5X](https://github.com/tnunnink/L5Sharp/blob/main/src/L5Sharp.Core/L5X.cs).

L5X contains all the root container elements such as Tag, Task, etc as you have pointed out. It's set up this is because not all L5X files are complete project exports. Some files may just be a component export such as Module or Program. In this case they don't always contains the root Controller element. Of course, you can navigate back up and down the tree by getting the reference to the parent L5X from Controller like you are doing. 

The compiler warning is because L5X is nullable. It has to be because you can create instance of component in memory. If you instantiate a new Tag like `var tag = new Tag("MyTag", 100);` then there is no "attached" L5X parent to retrieve. It doesn't get that reference until you add it to an existing L5X object via `l5X.Tags.Add(tag);` (or something similar). You can handle the compiler warning by adding a null propagation operator or checking the L5X to ensure it is not null before iterating. The warning is just telling you to make sure you handle any null reference cases.

Let me know if this answers your question or not.

tnunnink, 2024-12-10:
`public IEnumerable<Tag> Tags => L5X?.Tags ?? [];`

This would let you write:

```
foreach (tag in controller.Tags)
{
    //do stuff
}
```

YouCanCallMeGeorge, 2024-12-10:
I understand that some L5X files are for subsets of a controller; I regularly export programs and AOIs as L5X files for distribution to other programmers.

I'm not clear as to why the L5Sharp.Core.Controller class does not have a Tags member, while the L5Sharp.Core.Program class does have a Tags member, even though not all programs have Tags. (It happens that the ACD file I chose for initial testing contains a program that has no tags. Interestingly, the foreach statement handled that situation without throwing an exception.)

If it's a simple matter to add Tasks and Tags to the L5Sharp.Core.Controller class, that would be great, simplifying the flow of object handles through the tools that we will be developing.

I am really impressed with the completeness and precision of the work you have done. I hesitated to be so picky...

As a side note, back in the '90s when all my programs were written in C or sometimes assembly language, every C function began with tests for null pointers. I've become somewhat complacent these days since none of my high level language programs run in real time, and I create all the objects that are used. But your point is well taken that in this environment, when working with code created by others, I cannot count on the existence of any particular component in an L5X file.

tnunnink, 2024-12-10:
I appreciate the feedback! It's always good to hear how people are using this library and pain points they encounter. I've basically developed the entire library with very little/no collaboration, so I think is suffers from some differing points of view, so thanks for providing one!

I agree that from a conceptual standpoint having Tags in Controller, similar to Program, make sense. I will go ahead and add these helper properties to Controller for now and get that released in the next version.

My reasoning again was mostly based on how the L5X was structured and how it can serialize differently depending on what you export. Since sometimes the L5X serializes without a top-level controller, I opted to put the Tags collection in the root L5X class, similar to Tasks, Modules, Programs, etc. for consistency. Maybe this wasn't the best approach - it's hard to say. I think it will only ever export controller tag if you export the entire project anyway, so that could be one way of looking at it that I didn't originally consider.

YouCanCallMeGeorge, 2024-12-10:
Another way to think of it is that nearly every controller will have Tags and Tasks. So if the Controller class is used in a given L5X file, then Controller.Tasks and Controller.Tags are very likely to be used.

Thanks so much for your quick and cooperative response.

tnunnink, 2024-12-10:
Okay [4.7.3](https://www.nuget.org/packages/L5Sharp/4.7.3) adds basically all the same components collections to Controller, including Tags and Tasks. Check this out and let me know if I missed anything.

YouCanCallMeGeorge, 2024-12-10:
AWESOME!!!
Thank you.

closed 2025-01-10 by tnunnink