tnunnink/L5Sharp #52: Bit reference tag in NeutralText.Instructions() causes index out of bounds

controlaware, 2025-03-31:
The error actually occurs in TagName.IsQualifiedName, when the rung includes a tag that uses bit addressing the members list has a blank member, causing char.IsLetter to throw out of bounds. 

To recreate, make a DINT tag called DintTest and one called Offset, and use a rung XIC(DintTest.[Offset]);

Call rung.Text.Instructions(), this will throw the out of range error.

tnunnink, 2025-03-31:
Check out the latest version to see if that fixes it for you. Let me know if you still have issues. Thanks for reporting this!

closed 2025-03-31 by tnunnink

controlaware, 2025-04-01:
Thanks, I'll give it a try, I found another bug I'll report in a new issue that prevents me from using the Nuget package at the moment.