tnunnink/L5Sharp #56: handle RackConnection and implicit alias tags

crmoore, 2025-04-09:
When using rack optimized connections with remote io, logix creates alias tags implicitly to the parent module.

In the l5x, RackConnection is a possible sub element of Connections. It may have InAliasTag and OutAliasTag elements, and Comments.

When those elements are present, alias tags should be created pointing at the parent module's in and out tags.

tnunnink, 2025-04-09:
Are you able to provide a snippet of what the RackConnection element looks like? Not seeing anything in the manual.

crmoore, 2025-04-09:
The manual "Logix 5000 controllers import/export" mentions it with nearly no details.

Mine is pretty empty.

`<Connections>
<RackConnection>
<InAliasTag/>
<OutAliasTag/>
</RackConnection>
</Connections>
`

tnunnink, 2025-04-10:
Thanks! I'll have to dig into this more when I get time.

May I ask the catalog number of the module that is producing that rack connection element so that I can experiment with it?

tnunnink, 2025-04-10:
I see, so the actual tag data is contained under the parent module IO tag elements. So we would use the slot number and parent module name to lookup the corresponding tag data in the parent module. 

And you want to be able to query these alias tags and have it return a Tag instance with that data populated? Am I understanding your request correctly? 

This might be tricky to implement. I'll have to think through it.

crmoore, 2025-04-11:
Even if it returned some identifier that I could manually de-reference, it would be useful.

tnunnink, 2025-05-05:
Added support for this in 5.1.0. You should be able to query the In/Out alias tag same as other module In/Out tags. When a Tag element is deserialized we are checking the element name and if it is an alias tag it will retrieve the corresponding structure member from the underlying L5X document. This will of course rely on that content being available. Let me know if you have any issues.

closed 2025-05-05 by tnunnink