tnunnink/L5Sharp #57: Flex 5000 Safety IO Modules

Ryushi5, 2025-04-30:
One issue that I've been trying to figure out is how to create Flex 500 safety modules that will import into Studio 5000. From what I can see, the 5094 safety modules have a "Connections" child that details a lot of information that is difficult to programmatically generate.

For example from a 5094-OB16S:

`<Connections>
<Connection Name="SafetyInput" RPI="20000" Type="SafetyInputDataDriven" InputSize="68" EventID="0" ProgrammaticallySendEventTrigger="false" TimeoutMultiplier="2" NetworkDelayMultiplier="200" ReactionTimeLimit="80" MaxObservedNetworkDelay="0" Priority="High"
 InputConnectionType="Unicast" InputProductionTrigger="Application" ConnectionPath="20 04 24 b5 20 04 24 c7 20 04 25 00 50 03" InputTagSuffix="I">
<InputTag ExternalAccess="Read/Write" OpcUaAccess="None">
<Data Format="Decorated">
<Structure DataType="AB:5000_SDO16:I:0">
<DataValueMember Name="RunMode" DataType="BOOL" Value="0"/>
<DataValueMember Name="ConnectionFaulted" DataType="BOOL" Value="1"/>
<DataValueMember Name="DiagnosticActive" DataType="BOOL" Value="0"/>
<DataValueMember Name="DiagnosticSequenceCount" DataType="SINT" Radix="Decimal" Value="0"/>
<StructureMember Name="Pt00" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt01" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt02" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt03" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt04" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt05" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt06" DataType="AB:5000_SafetyReadback_Channel:I:0">
<DataValueMember Name="Readback" DataType="BOOL" Value="0"/>
<DataValueMember Name="Fault" DataType="BOOL" Value="1"/>
<DataValueMember Name="Uncertain" DataType="BOOL" Value="0"/>
<DataValueMember Name="ShortCircuit" DataType="BOOL" Value="0"/>
<DataValueMember Name="FieldPowerOff" DataType="BOOL" Value="0"/>
<DataValueMember Name="Status" DataType="BOOL" Value="0"/>
</StructureMember>
[... 130 lines cut ...]
<DataValueMember Name="ResetFault" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt12" DataType="CHANNEL_SDO:O:0">
<DataValueMember Name="Data" DataType="BOOL" Value="0"/>
<DataValueMember Name="ResetFault" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt13" DataType="CHANNEL_SDO:O:0">
<DataValueMember Name="Data" DataType="BOOL" Value="0"/>
<DataValueMember Name="ResetFault" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt14" DataType="CHANNEL_SDO:O:0">
<DataValueMember Name="Data" DataType="BOOL" Value="0"/>
<DataValueMember Name="ResetFault" DataType="BOOL" Value="0"/>
</StructureMember>
<StructureMember Name="Pt15" DataType="CHANNEL_SDO:O:0">
<DataValueMember Name="Data" DataType="BOOL" Value="0"/>
<DataValueMember Name="ResetFault" DataType="BOOL" Value="0"/>
</StructureMember>
</Structure>
</Data>
</OutputTag>
</Connection>
</Connections>`

I just wanted to see if there were any thoughts on being able to generate these modules.

Ryushi5, 2025-04-30:
I'm sorry, I didn't realize this was a duplicate of #45

Ryushi5, 2025-04-30:
From what I can tell, it appears that the data is required for import. I've even tried replacing all digits with "0"s to see if the data needs to be valid: and it errors out. So I'm not sure what Rockwell is doing, but they're not making things easy.

tnunnink, 2025-04-30:
Well that stinks. I'll play around with it and see if I can figure something out, but that doesn't sound promising.

tnunnink, 2025-04-30:
Thanks for the info. I was able to confirm what you are seeing. 

Since we can't omit the Data element and we have no way of generating it (seems internally dependent on the module type) then I'm not sure what to do here. I looked to see if that data was stored in the Rockwell module catalog xml file but couldn't find anything. I'm not even sure what the SafetyScript is representing from Studio interface. Seems to be internal to the software and safety devices.

I did find that there are some data values in the array that is does not care about. I took data from a similar module type and pasted into another, and it still imported fine, but there was some overlap in values, so there are certainly some values that are somehow tied to the device type and need to be correct.

My only thought at this time is that we could first add SafetyScript as an element/property to the Connections object so that we could at least read this from existing L5X files that are exported from Studio. You could then in theory create new instances from an existing exported module template, update properties as needed, add to new L5X project file, and then import without error.  

The API could probably be improved here too. Maybe we should add the ability to instantiate or add a Module from an L5X file directly, so to avoid having to load file, find module, duplicate it, updated it, then add to another L5X. Some sort of factory method/class. Not sure how that would look.

I open to other ideas. Whatever can help improve the workflow is fine by me.

tnunnink, 2025-05-01:
After thinking more, we don't even need to add SafetyScript property to existing objects. The element should get "deserialized" regardless. The property would just be an accessor to an element that no one would know what to do with.

Here is what I have in mind for the factory method to create Modules form existing L5X templates.

```csharp
//File path to known L5X containing module for catalog you are trying to create
var template = Known.Test;
        
//Takes template file, catalog number, and config delegate to update whatever properties you need
var module = Module.Create(template, "1756-EN2T", m =>
{
    m.Name = "NewName";
    m.Revision = 1.23;
    m.Description = "This is a new module instance based on an existing template";
    m.Inhibited = true;
});
```

This is similar to the current `Create` factory method when you think about it. Just instead of using the Rockwell predefined ModuleCatalog file you are you using a self-provided L5X file. The only downside is that the onus is on consumers to have that content readily available. But with this you could at least create new safety enabled modules that will import successfully without having to manually update the XML or add some custom code to do so.

Ryushi5, 2025-05-01:
Yeah, I think that would work well. I already do something similar for creating third party modules, and having a more streamlined way of doing it would be very welcome. Thank you for looking into this!

closed 2025-05-04 by tnunnink