tnunnink/L5Sharp #59: Add link to auto-generated documentation from DeepWiki

andy-gxpaasio, 2025-05-07:
Last night I autogenerated a doc package using a tool called DeepWiki.  While I can't speak to its absolutely accuracy the depth and detail and incredible.   

At the top it says 

> Add a link to this wiki in the repository's README file to auto refresh the wiki weekly with the latest code changes.

I figured I'd raise an issue instead of proposing a spot in your ReadMe via a PR

https://deepwiki.com/tnunnink/L5Sharp

tnunnink, 2025-05-07:
Yeah this is cool thanks for sharing. It does a good job of organizing the content and breaking it all down in a coherent way. I have not read through all of it yet but in general it seems to be fairly accurate. It is interesting to see the things it focuses on.

I do see that a some of this is just taken directly from the documentation I have created. It looks like some is coming from the docfx articles which are outdated, and that is probably the source of some error. I've been trying to decide on whether or not to update/maintain that or to remove it all together. This might convince me to remove it, but I'll have to think through it. 

Is this a paid tool? Are you able to prompt it to change wording or content, or to tell it which parts are incorrect? Or is it you get what you get?

andy-gxpaasio, 2025-05-07:
Right now it's free, but you get put in a queue if you ask for a repo that hasn't been indexed.   Last night when I put yours in I was 65 in queue and it indexed in maybe an hour or two.  This morning I put in another one and I was 1280 in queue.  Don't know if it's time of day or getting exponentially more popular.  However, I have also learned that there is a self hosted version at 
[https://github.com/AsyncFuncAI/deepwiki-open](https://github.com/AsyncFuncAI/deepwiki-open).  

The other interesting feature is that after it generated all of the docs I asked it a specific question about creating some extension methods to help me with exporting tag data to structured formats like CSV or TSV.   It used Devin to generate a good amount of code with examples.  Did I like the way it approached the problem, not really but it was neat to be able to build on the model understanding to write code against the library.

tnunnink, 2025-05-07:
Yeah that is really powerful. I just tested that out a bit and it did well generating answers with multiple code examples. I think I'm sold. If you want to submit a PR, go ahead, or I can add it in when I get time. Thanks for making this suggestion.

closed 2025-05-08 by tnunnink